EngineerGuy314/pico-WSPRer #1: Grid Co-Ordinates  not as expected

sivaelid, 2024-03-31:
Having  used the V2 Code, I find that the reported grid is incorect. I have check several GPS receivers and my QTH is IO81wk49 with co-ordinates of    lat   51.4559367°    Lng          -2.1259093°

Using the code I had  displayed at the same l,ocation   lat 512736246  and long  -20754834 .  This showed  a  grid of IO 81xg so It seems to have some error in 5th and 6th character. The Square is  5 grids south and 1 grid east of the actual square .

I hope this  helps Simon G8HAM

EngineerGuy314, 2024-04-06:
I had  noticed something similar, and assumed that the raw data coming from the inexpensive GPS was off. But I will investigate further, thank you for the comment.

sivaelid, 2024-04-06:
Hi I have been looking to day at the issue, and how Co-ordinates to maidenhead is done . I have swapped 3 Different makes of GPS and with different NEMA Message  $GPRMC and $GNRMC. All give the correct co-ordinates but are  always giving the incorrect Grid. I have come to the conclusion so far that It may be  the rounding down element  of the sub-Square element that  is incorrect. If you add or subtract the  1 to the se values you can make it correct . Still looking If I find issue  then will let you know.  PS some great work so far . I would say that looking on wspr with the current build I am  not getting manay sport comare to the normal ones. Frequency Ok but ? changes in coding of message. Any Way that for your creative thought and skills Simon G8HAM  Sent from Mail for Windows From: EngineerGuy314Sent: 06 April 2024 15:10To: EngineerGuy314/pico-WSPRerCc: Si; AuthorSubject: Re: [EngineerGuy314/pico-WSPRer] Grid Co-Ordinates not as expected (Issue #1) I had noticed something similar, and assumed that the raw data coming from the inexpensive GPS was off. But I will investigate further, thank you for the comment.

EngineerGuy314, 2024-04-06:
bug found and corrected! the gps parser code i started with assumed lat/lon were sent as decimal, when in fact they come as that stupid degree/minute.minute format.

closed 2024-04-06 by EngineerGuy314

sivaelid, 2024-04-06:
Brill Well done Sent from Mail for Windows From: EngineerGuy314Sent: 06 April 2024 17:17To: EngineerGuy314/pico-WSPRerCc: Si; AuthorSubject: Re: [EngineerGuy314/pico-WSPRer] Grid Co-Ordinates not as expected (Issue #1) bug found and corrected! the gps parser code i started with assumed lat/lon were sent as decimal, when in fact they come as that stupid degree/minute.minute format.