EngineerGuy314/pico-WSPRer #2: Strange  activity in the RF transmission

sivaelid, 2024-03-31:


sivaelid, 2024-03-31:
The RF frequency when the   "d_force_xmit_for_testing  set to  YES  the  transmission is correct output of 14.097060 but when  the setting is set to "NO"  then the  transmission on the every 10 minute interval is at  13.091718 MHz.. I  Have check this with a frequency counter and with an HF rig to check  the centre of the transmission spectrum. Very odd any one else seen this ?

EngineerGuy314, 2024-04-06:
Hmm, I think I had seen something similar (several revisions back) but not recently.

sivaelid, 2024-04-06:
Hi this seems solve in the latest versions so Closed

closed 2024-04-06 by sivaelid

sivaelid, 2024-04-06:
Hi Sorry To Ask probably a daft  Question, but where does the altitude message come from as the Message type $GNRMC does not contain  it. I  only ask as in my case seems to be  reporting about twice the height but often does move as antenna is indoors. Just wondered if a different message type also used ? Thanks Simon G8HAM

sivaelid, 2024-04-06:
Hi Just to say I put the Pi Pico on frequency counter tonight  and the  problem of frequency jumping  has   just appeared again. Goes from 14.09760 mHz( lane 2 )  to 13.09193 mHz but is moving around a lot .It look like lost PLL lock and jumps back for a second or two to the  correct frequency and rock steady on the correct frequency   10-20s and then back to the 13.09193mHz  area. Found the  Altitude Message being used as  $GNGGA so look good but must have very poor VDOP. Simon Sent from Mail for Windows From: EngineerGuy314Sent: 06 April 2024 17:21To: EngineerGuy314/pico-WSPRerCc: Si; AuthorSubject: Re: [EngineerGuy314/pico-WSPRer] Strange activity in the RF transmission (Issue #2) Hmm, I think I had seen something similar (several revisions back) but not recently.

EngineerGuy314, 2024-04-08:
yes, altitude comes from GNGGA. but the altitude value is definitely very crude, i see it jump around +-40 meters. My understanding is that vertical positioning of GPS is always very inacurate when you have few satellites (such as when testing indoors).  But my hope is that when floating there will be vastly superior gps reception, and hence more accurate altitude reports.