EngineerGuy314/pico-WSPRer #4: maidenhead.c  needs Grad Seconds Miniutes conversation to decimal

whallmann, 2024-04-12:
The RMC sentence delivers coordinates used like in APRS.  So i wonder why the QRA locator shows not exact the right one.
The maidenhead function :
char* get_mh(double lat, double lon, int size) 
needs an converting like this:

> double gms2dec(double gms) {
> 	double degrees;
> 	double minutes;
>        degrees = (int)(gms);
>        minutes = (gms - degrees) * 100;
>        degrees = degrees + minutes/60;
>     return degrees;
> }

and insert the two lines:

> 
> char* get_mh(double lat, double lon, int size) {
>     static char locator[11];
>     double LON_F[]={20,2.0,0.083333,0.008333,0.0003472083333333333};
>     double LAT_F[]={10,1.0,0.0416665,0.004166,0.0001735833333333333};
>     int i;
>     // if lat, lon in APRS format then convert to dezimal
>     **lat = gms2dec(lat);**
>     **lon = gms2dec(lon);**

I think the RMC values are always the same format over all GPS chips. So it was here with two types of ublox units.
Please check it with your type of GPS. 

Thanks
Wolf

EngineerGuy314, 2024-04-12:
so my understanding is that the GNRMC format gives latitude as ddmm.mmmmmmm. The code on lines 213 and 228 of gpstime.c is supposed to convert it to pure degrees. (Or at least pure degrees multiplied by large value to keep the format 64bit integer...)

The earlier versions of this code (including the one that flew around the world) was not doing this correctly. It was only the commit I made on April 6, 2024 that fixed this problem.

At least I *think* it fixed the problem. The code on lines 213 and 228 is not very clear or elegant, but it was my intention that it is doing essentially the same calculation (minutes = (gms - degrees) * 100; degrees = degrees + minutes/60;) that you showed in your comment.

If you think its incorrect please let me know, but my bench testing currently is showing accurate values.

whallmann, 2024-04-16:
An i would knowledge that i checked this with the last version. It is fixed as told.

closed 2024-04-16 by whallmann