EngineerGuy314/pico-WSPRer #9: tranquito shows now Altitude and Speed

whallmann, 2024-05-09:
My first ballon DF7PN with channel 493 is flying. I wonder why i see no Altitude and Speed.
I thought the EncPower is the aprox. Height level. 
Is it posible to calculate the Altitude anyway?
Or is the implemantation of the U4B protocol not complete?
See here: https://traquito.github.io/search/spots/dashboard/?band=20m&channel=493&callsign=DF7PN&dtGte=2024-04-09&dtLte=
73 Wolf DF7PN

EngineerGuy314, 2024-05-09:
i looked at your data, and I don't know how to explain that.
>>EncPower is the aprox. Height level.

that is only true for the Zachtek protocol, it looks like you are tracking it with U4B protocol

Did you use the latest version of the software?

In a previous version of the software if it loses GPS lock it will still transmit, and use the previous known location, but put in a zero for altitude. but that doesnt seem to be whats going on here

EngineerGuy314, 2024-05-11:
i see you have the altitude reporting now, i will close this. Also want to explain the knots your seeing. The tracking websites usually figure out your speed pretty well based on change in position. So i am sneaking the number of satellites seen by the GPS into the knots field (the tracking websites multiply this by two unfortunately).  So basically that last spot where it reported 20 knots means your gps receiver was viewing 10 satellites.
[image]

closed 2024-05-11 by EngineerGuy314