EngineerGuy314/pico-WSPRer #13: U4B shows wrong  Locatorprefix

whallmann, 2024-05-11:
I watched on the console with verbose 5 a strange thing .
The locator is some kind of mismatch. See hardcopy please
[image]

that may be the reason why not the correct locators are received at
https://traquito.github.io/search/spots/dashboard/?band=20m&channel=493&callsign=DF7PN&dtGte=2024-03-24&dtLte=
[image]

EngineerGuy314, 2024-05-11:
interesting.  One question, when you saw the terminal show JN49u?? was that the same transmission that the website recorded as BB00AV? or was that a different occurence?

whallmann, 2024-05-11:
Oh, dont remember excactly.
If i look to the traquita site. all my transmissions are same bad locator
today. I only saw this one and made foto. maybe this bad line was steady.
cant check it now because auf little fieldday.
making aurora dx qsos on 2 mtr.

73 wolf

EngineerGuy314, 2024-05-11:
OK, i am about 90% sure i fixed it (forgot to null terminate a string).  I will close this issue, please feel free to open a new one if still no good. The link and screenshots you post are extremely helpful in tracking down these gremlins!

closed 2024-05-11 by EngineerGuy314

whallmann, 2024-05-11:
Yes Rob. Thanks to you and good bye to the gremlins 😁