EngineerGuy314/pico-WSPRer #14: GPS fix, but getting wrong grid square.

WildMany, 2024-05-11:
Tried on laptop, tried on battery (power bank), and during two hours, not a single correct gridsquare (JN95):
[image]

Meanwhile, during this period assembled ICT tracker:
[image]
[image]
[image]

RPi Pico tracker should be easy and fast to set, but it is not so easy.

73, Milan, 9A4GE.

EngineerGuy314, 2024-05-11:
this may have been fixed by the change I just pushed for Whallmann. if the current version still doesnt work please turn verbosity to 5 and try to capture some of the debug messages.

WildMany, 2024-05-11:
Now seems that it working just fine: JN95JG. Will test in next few hours.

73, Milan, 9A4GE

closed 2024-05-12 by EngineerGuy314