EngineerGuy314/pico-WSPRer #15: Traquito shows values higher then 39 deg Celsius as negativ values

whallmann, 2024-05-12:
A small thing was watched in the table of traquito
https://traquito.github.io/search/spots/dashboard/?band=20m&channel=493&callsign=DF7PN&dtGte=2024-03-24&dtLte=
[1] If there are TEMPC values over 39 then the value gets negative.
[2] Also KNOTS value witch includes the double of SATS went to very high values that cant be real.
maybe there is a solution for this. Thanks 
73 Wolf
[image]

EngineerGuy314, 2024-05-12:
I saw the same thing with the negative temps over 39. I think that was a limitation of the protocol and the way the tracking websites display the data. Wasn't anything I could do about it, and it should never get that hot up in the stratosphere anyway! 

That problem with the knots is strange though. I need to look into that.

whallmann, 2024-05-12:
Ok Rob, thanks for looking to it.
I thought it could be a transformation issue or something like this.
The values are definitly out of practis.
73 Wolf.

ps: tested around the house today. Looks very well. Ready to prepare the
next balloon.

closed 2024-05-14 by EngineerGuy314