EngineerGuy314/pico-WSPRer #19: can't run clock speed below 110Mhz

EngineerGuy314, 2024-06-02:
serych's latest contribution optimized pco code so that everything works great at 115Mhz pico clock speed (or faster). 

But at 110Mhz for instance the timing for packet transmission is incorrect, even though oscillator output frequency is still correct. 

documenting this as an issue so we can track anything we find.

I might need to remove some of the print statements used at higher verbosity settings....

EngineerGuy314, 2024-06-03:
Hmm, I just ran a test overnight at 110Mhz and full verbosity=9. Packets seem to be coming through and scheduled just fine.

closed 2024-06-03 by EngineerGuy314

serych, 2024-06-03:
Perfect! And I think I found the problem with out of schedule transmission. Function WSPRbeaconInit called on line 81 of main.c sets the schedule hardcoded to minutes 1, 2 and 3 only according to the type of protocol requested in id13. The actual scheduler is called later in the for (;;) loop. 
So it's possible that when WSPRer is started around minute 0, the transmission will occur according to that initialization. 
I haven't fixed it yet, but I think that's the source of the problem.

serych, 2024-06-03:
**Ignore my last comment WSPRbeaconInit is ok. I just badly understood it. But it is defined in two places (main.c and WSPRbeacon.c)
I'll analyse it further...**

serych, 2024-06-04:
So today it is working like a charm. WSJT-X log file:
```
240604 1736  29 -0.10  14.0971054  OK1SE JO70 10           0  0.62  1  1    0  0   0     1   782
240604 1738  30  0.02  14.0971054  Q50VSA ML87 20          0  0.70  1  1    0  0   0     1   802
240604 1746  28 -0.06  14.0971054  OK1SE JO70 10           0  0.92  1  1    0  0   0     1   810
240604 1748  29  0.11  14.0971054  Q50VTI ML86 10          0  0.92  1  1    0  0   0     1   810
240604 1756  28  0.02  14.0971056  OK1SE JO70 10           0  0.93  1  1    0  0   0     1   810
240604 1758  27  0.19  14.0971059  Q50VTI ML86 10          0  0.88  1  1    0  0   0     1   810
240604 1806  27  0.28  14.0971057  OK1SE JO70 10           0  0.78  1  1    0  0   0     1   767
240604 1808  28  0.45  14.0971057  Q50VSA ML87 47          0  0.73  1  1    0  0   0     1   779
240604 1816  27  0.37  14.0971057  OK1SE JO70 10           0  0.90  1  1    0  0   0     1   810
240604 1818  28  0.54  14.0971059  Q50VRT MI23 40          0  0.90  1  1    0  0   0     1   810
240604 1826  27 -0.23  14.0971059  OK1SE JO70 10           0  0.90  1  1    0  0   0     1   810
240604 1828  27  0.58  14.0971061  Q50VRT MI32 30          0  0.90  1  1    0  0   0     1   810
240604 1836  28 -0.19  14.0971061  OK1SE JO70 10           0  0.93  1  1    0  0   0     1   810
240604 1838  28 -0.02  14.0971063  Q50VRT MI32 30          0  0.89  1  1    0  0   0     1   810
---------------------------------------------------------------------------------------------------
240604 1906  27 -0.02  14.0971055  OK1SE JO70 10           0  0.74  1  1    0  0   0     1   766
240604 1908  28  0.15  14.0971058  Q50VRW MI33 27          0  0.73  1  1    0  0   0     1   773
``` 
I have rewriten all the menu settings before the test and may be it's the reason why it was not working in the past days. It showed correct values on the screen, but may be there were some other residues in the settings strings (from time to time some setting is changed after many reloads of .u2f to pico). 
Above the line verbosity was set to 1, below the line I set it to 9 to have the same settings as yesterday when it was misscheduling packets.  All seems to be OK.