EngineerGuy314/pico-WSPRer #26: new launch, good results, but is there room for improvement on drift?

EngineerGuy314, 2024-06-14:
Successful flight test of software as of June 13 2024, using 6 solar cells. Validated operation without an external MOSFET (3 IO in parallel low-side driving GPS module instead). Survived the first night, and woke up at 33 degree solar angle. Not great, but pretty good for The Cheapest Tracker In The World™. 

But it is not receiving 100% spots, and still getting some reports with -4 drift. I am confident it has 100% gps reception, and the pico is mounted out of direct sunlight.

Is this as good as is possible with this architecture? Any room for improvement in the software? Or should I try encasing the tracker in thermal insulation as others have done?
[image]
[image]

serych, 2024-06-14:
Great!!! Which CPU frequency is that balloon running at?

serych, 2024-06-16:
On the pilgrimage to Santiago de Compostella? :-)

EngineerGuy314, 2024-06-17:
I had it set for the new default, 115Mhz.

EngineerGuy314, 2024-06-17:
I think i know whats going on. I have seen this flight report some very extreme temperature fluctutations.  from 8 degree C to 36 degree C in less than an hour and a half.  Looking at weather maps for that time and location, it was over the ocean and crossed from an area of no clouds to thick, high clouds. It was still flying above the clouds, but the reflected sunlight must be bright enough to heat it up significantly. I did some bench testing, and there is no way that the current GPS compensation code can maintain the output frequency within the narrow drift window that WSPR uses when temperature changes that rapidly.

There is no easy fix to this.  When trying to measure the frequency error by counting microseconds in between the PPS pulses from the GPS the immediate results are very crude and jumpy (if there was a reliable way to count nanoseconds instead of microseconds, this wouldn't be an issue).  So Roman's original code takes the average of the microsecond counts over the previous 32 seconds. This gives smooth and precise frequency compensation, but with an unavoidable lag.  

I'm working on an idea that uses the onboard temperature sensor to implement a software based TCXO. I have this running on the bench (with the GPS compensation completely disabled) and it works remarkably well. But I need to work out a way that it still uses the GPS for long-term frequency compensation and offset, but does short term response based on temperature.  

My concern is that within the narrow range of temps I can simulate on my workbench the temperature/frequency relationship is perfectly linear. But i dont know how the xtal on the pico responds at negative temperatures. 

of course, the easiest solution might just be to put some shielding under the board to block the light reflected from the clouds. Temperature will still fluctuate, but not nearly as rapidly.

serych, 2024-06-17:
First problem of temperature measurements on pico is the Uref for ADC. It is directly connected to power supply on pico and is strongly fluctuating (it is even worse when powered by non stable solar power supply). I started development of the pico-HABakuk board (https://github.com/serych/pico-HABakuk) on which I will use LM4040 voltage reference. And I think also calculating the floating average of several temperature measurements could help. On my first launch I want to put the board into the piece of hollowed extruded polystyren construction thermal insulation. It is very lightweight and has good thermal insulation properties.

EngineerGuy314, 2024-06-17:
I did put a simple IIR lowpass filter on the temperature value, because i noticed random fluctuating. I was shocked it worked as well as it did.

EngineerGuy314, 2024-06-19:
added a selectable option to use temperature compensation. Works good on the bench, will need a flight test.

closed 2024-06-19 by EngineerGuy314