EngineerGuy314/pico-WSPRer #27: Option to use as a stationary WSPR beacon?

nilspupils, 2024-06-28:
First of all thank you very much for this wonderful piece of software! 
Is there any chance you are planning to develop this also into a stationary beacon with the option to disable tracking and to choose the transmitting band (and processor clock) within the console options? 
Thanks and best wishes, 
Nils

serych, 2024-06-28:
It should be possible to add the CPU clock option to the console menu. 
What do you mean by disabling tracking? There are Maidenhead coordinates in the normal WSPR packet and you need to use GPS receiver to correct the oscillator frequency. Or do you just mean a beacon transmitting constant frequency? You still need a GPS to correct it. 

Jakub, OK1SE

EngineerGuy314, 2024-06-28:
> into a stationary beacon with the option to disable tracking
as serych said there is no reason you cannot use it as stationary beacon right now.

If you configure it just for the Zachtek protocol, it will transmit perfectly standard WSPR Type-3 messages. oops, i take that back, it will encode altitude into the power fields, which is definitely not standard...

Adding a config option to do just a single, WSPR type-1 beacon message shouldnt be too hard, and I agree is good idea. I will keep this issue open until I can get around to it.

But I guess the question is, how flexible does it need to be? Does it need option to specify which timeslot? Every 2 minutes or once ten minutes enough? At specific fixed frequencies, or frequency hopping? Alternating different bands on different timeslots? I might need to re-think the way I am doing the user-interface,  it might be very cumbersome to implement this many selections.  But I can see a lot of use for a dirt-cheap WSPR beacon.

30, 40 or 80 meter bands shouldn't be too hard to implement.  And for a stationary application with an AC power source we could put back the overclocking and probably run it on higher frequencies too.

EngineerGuy314, 2024-06-28:
>> You still need a GPS to correct it.
absolutely. Although, if you use a pico-W, it might be possible to do frequency correction based on reading time values from ntp.org or something similar. It would have to average out readings over several minutes, but that would be enough if you mount it somewhere with relatively constant temperature. not really something i want to pursue.

whallmann, 2024-06-28:
my opinion:  dont put some new bugs in there for this tiny ballon version.
It may be a better idea to Clone a new project?!

EngineerGuy314, 2024-07-03:
I will add these to the list of features for a future cloned or forked version.

closed 2024-07-03 by EngineerGuy314