EngineerGuy314/pico-WSPRer #35: maidenhead grid values still sometimes wrong.. stumped..

EngineerGuy314, 2024-08-13:
I launched [this flight](https://traquito.github.io/search/spots/dashboard/?band=20m&callsign=KC3LBR&dtGte=2024-08-12&dtLte=&channel=315) today with the latest code, which i thought had fixed the issue, but apparently not.
[image]
there is a huge and impossible jump in the middle of a string of smooth position reports. Basically when the 5th grid char rolled-over from a U to an A, the 3rd grid character did not roll over from 8 to 9. not right away anyway.

the get_mh() function in maidenhead.c has a bunch of division and fmod instructions, and I think that when some values of latitude or latitude result in division values that are very very close but not quite an integer ( ie x.99999999999999999999999...) is when this problem occurs. it floors when it should round, or vice versa or something.

But I keep comparing the code in maidenhead.c to information online on how to calculate maidenhead, and it seems to be correct. Originally it had some incorrectly rounded values in the LON_F and LAT_F arrays, but i fixed those previously.

Any ideas? I'm stumped.

EngineerGuy314, 2024-08-13:
oh crap, just discovered a major flaw in my thinking.  The code is constantly getting new GPS coordinates, and constantly re-calculating new maidenhead grids from it.

The first U4B packet encodes the first 4 grid chars, and the 2nd U4B packet encodes the 5th and 6th grid chars.

The problem is that during the two minutes in between creating the 1st and 2nd U4B packets, the balloon is traveling and the grid chars are changing. I am encoding the 1st and 2nd packets from potentially two different sets of maidenhead grids.

Simple fix: When I calculate the 6 char rid location to encode the 1st U4B packet, I need to "freeze" that grid location and use that same grid location when encoding the 2nd U4B packet. There is no need to be constantly re-calculating the grid location, it only needs to be done once every ten minutes, right before encoding the 1st packet. its so obvious, now.... lol

whallmann, 2024-08-23:
Congrats, it sound easy. Is there a new Version for testing?
cu Wolf

EngineerGuy314, 2024-08-24:
yes it was an easy fix,  it just took me a while to get around to it. the version I pushed today has the fix.

closed 2024-08-24 by EngineerGuy314