EngineerGuy314/pico-WSPRer #42: Schematics: antenna connection

IU5HKU, 2024-10-15:
First and foremost, thank you for this great piece of code, this isn't an issue but a question. I wanna connect the antenna's pins to an 1:9 balun, to playing with various length of wire. I know, isn't the best suitable antenna for this project, a dipole could be way better...but: do i need some decoupling capacitors between antenna output and the balun, who de facto short-circuits the output pins? There is a risk to burn the output? If yes (and i think this will be the answer) can you suggest a proper value? Thank you again, have a nice day!

EngineerGuy314, 2024-10-15:
because of limitations in the way the PIO code is written, after the first transmission, whenever it is idle in between transmissions one leg idles high, and the other low, so yes a balun would act as a short circuit to this DC voltage.

When testing at home I used a 1uF cap for DC blocking with very good results. 

But the RP2040 seems very tolerant to shorts. One person has been running a Pi Pico as a standalone beacon into a balun with no caps at all, and other than the pico running a "little warmer" there have been no ill effects.  But I definitely recommend the cap to be on the safe side...

closed 2024-10-15 by EngineerGuy314

IU5HKU, 2024-10-15:
Thank you so much for your quick reply, I'll go for a 1uF cap, I've already found it in my spare parts junkyard, I hope to test the whole thing ASAP