EngineerGuy314/pico-WSPRer #43: are the "build it yourself" instructions up to date?

knormoyle, 2024-10-21:
I was able to install/build the sdk

but when I tried pico-WSPRER build, I got the following
(minimally it seems like something understand your directory structuree i.e. /home/rob/pico-WSPERer)

I subsequently removed the  CMakeCache.txt in the build dir that the git clone created.

That helped (I guess before the build.sh the instructions should say to remove that, the first time after git clone?)
it got further then (still not complete but different error)

Originally results without that
#**********************

kevin@pc8c:~/pico_install$ **git clone https://github.com/EngineerGuy314/pico-WSPRer.git**
Cloning into 'pico-WSPRer'...
remote: Enumerating objects: 4762, done.
remote: Counting objects: 100% (863/863), done.
remote: Compressing objects: 100% (309/309), done.
remote: Total 4762 (delta 663), reused 588 (delta 547), pack-reused 3899 (from 1)
Receiving objects: 100% (4762/4762), 43.38 MiB | 3.81 MiB/s, done.
Resolving deltas: 100% (2775/2775), done.

kevin@pc8c:~/pico_install$ **cd pico-WSPer**
 
kevin@pc8c:~/pico_install/pico-WSPRer$ **./build.sh**
mkdir: cannot create directory ‘build’: File exists
fatal: No url found for submodule path 'build/_deps/picotool-src' in .gitmodules
fatal: No url found for submodule path 'build/_deps/picotool-src' in .gitmodules

CMake Error: The current CMakeCache.txt directory /home/kevin/pico_install/pico-WSPRer/build/CMakeCache.txt is different than the directory /home/rob/pico-WSPRer/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt
CMake Error: The source "/home/kevin/pico_install/pico-WSPRer/CMakeLists.txt" does not match the source "/home/rob/pico-WSPRer/CMakeLists.txt" used to generate cache.  Re-run cmake with a different source directory.
CMake Error: The source directory "/home/rob/pico-WSPRer" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** [Makefile:3067: cmake_check_build_system] Error 1

not sure what the fatal about picotool-src is. Maybe it just doesn't know where my sdk is

I did get picotool installed during the sdk build..here's a grep of it from my notes
maybe just a side effect of not finding my sdk

 After rm build/CMakeCache.txt, and ./build.sh again, I get this
Still od it complains about picotool
seems to fail on pioasm

kevin@pc8c:~/pico_install/pico-WSPRer$ ./build.sh
mkdir: cannot create directory ‘build’: File exists
fatal: No url found for submodule path 'build/_deps/picotool-src' in .gitmodules
fatal: No url found for submodule path 'build/_deps/picotool-src' in .gitmodules
PICO_SDK_PATH is /home/kevin/pico_install/pico-sdk
Target board (PICO_BOARD) is 'pico'.
Using board configuration from /home/kevin/pico_install/pico-sdk/src/boards/include/boards/pico.h
Pico Platform (PICO_PLATFORM) is 'rp2040'.
Build type is Release
TinyUSB available at /home/kevin/pico_install/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
BTstack available at /home/kevin/pico_install/pico-sdk/lib/btstack
cyw43-driver available at /home/kevin/pico_install/pico-sdk/lib/cyw43-driver
lwIP available at /home/kevin/pico_install/pico-sdk/lib/lwip
mbedtls available at /home/kevin/pico_install/pico-sdk/lib/mbedtls
-- Configuring done (0.2s)
-- Generating done (0.0s)
-- Build files have been written to: /home/kevin/pico_install/pico-WSPRer/build
[  1%] Built target bs2_default
[  3%] Built target bs2_default_padded_checksummed_asm
[  3%] Generating dco2.pio.h
pioasm-install/pioasm/pioasm: 1: cannot open 4: No such file
pioasm-install/pioasm/pioasm: 1: ����Q�7��faGNU8: not found
pioasm-install/pioasm/pioasm: 1: cannot open 
                                             : No such file
pioasm-install/pioasm/pioasm: 11: y
                                   : not found
pioasm-install/pioasm/pioasm: 11: �: not found
pioasm-install/pioasm/pioasm: 12: Syntax error: ")" unexpected
make[2]: *** [CMakeFiles/pico-WSPRer_dco2_pio_h.dir/build.make:73: dco2.pio.h] Error 2
make[1]: *** [CMakeFiles/Makefile2:2109: CMakeFiles/pico-WSPRer_dco2_pio_h.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

knormoyle, 2024-10-21:
building on Ubuntu 24.04.1 LTS

knormoyle, 2024-10-21:
Ah, I think I found the issue
you have to run .**/clean.sh** before you do **build.sh**
that may be convention I'm not aware of, but should mention it in build instructions.

going much further now.

EngineerGuy314, 2024-10-22:
Before you tried the initial build, it looks like you skipped steps 1b and 1c:
[image]

I admit I don't know the make-file system very well, and what I do know I definitely do not like! There is definitely ways I can get it screwed up and need to start from the beginning. There is surely a better/easier way to set it up, but my intention is that if the build steps I listed are followed exactly as listed, it will work.

I will try it again with a fresh system, and update the instructions as needed. Until then, if you could please try:
-completely delete the pico-WSPRer folder
-do steps 1b and 1c
-clone pico-WSPRer and try to build it (without running clean.sh or changing anything).

please let me know if doing it that way works.

EngineerGuy314, 2024-10-23:
I updated the build instructions to be more thorough, and verified it works on a fresh install when following the new instructions exactly.

closed 2024-10-23 by EngineerGuy314

knormoyle, 2024-10-23:
thanks. I think I had done the "pico extras" and git submodule init as stated

the key for me seemed to be doing "clean.sh" then the "build.sh"

thanks. I think I'm good now. I'll check out your updated instructions