EngineerGuy314/pico-WSPRer #45: better config hand holding with messaging

knormoyle, 2024-10-24:
I just got my jlcpcb boards and they are great.

I loaded firmware and got the repeated messaging about no communications from gps or interrupted serial.

Interestingly this is the same repeated error message you get if you load a bare pico with the firmware and have no gps chip.

Yes the answer is obvious, but to help us people new to this tracker:

The messaging should give a hint about the need to set a config bit if you have the jlpcb board.

In the config help:

custom PCB IO mappings: Set to zero if you assemble the tracker from Pi Pico and separate GPS module, per the schematics. Set to '1' if using a custom PCB based on the gerber files in /PCB -TELEN config: (see section on TELEN)

Yes, one should always read the instructions, but it does make sense that it should behave in a way that tells you whats wrong if you load the firmware and it doesn't work right

Ideally, you could detect the existence of the jlpcb board somehow, and default the config correctly.
Minimally, just enhance the error message you get about gps communications when the config is wrong, to suggest setting the custom PCB mapping bit, if this is a jlcpcb board.

EngineerGuy314, 2024-10-24:
I tried to not add too much text on the parameter descriptions, because then it might wrap on some terminals and makes things look very confusing. 

There is that brief paragraph of info that gets displayed before going into the setup-screen proper. What if I add a big conspicuous note in there about the custom-PCB bit?

EngineerGuy314, 2024-11-03:
>>What if I add a big conspicuous note in there about the custom-PCB bit?
Done

closed 2024-11-03 by EngineerGuy314