EngineerGuy314/pico-WSPRer #46: erroneous 8640 meter altitude and BL sub grid

EngineerGuy314, 2024-10-27:
latest flight shows some outliers. [here](https://traquito.github.io/search/spots/dashboard/?band=20m&callsign=KC3LBR&dtGte=2024-10-06&dtLte=&channel=226)

When this occurs, the the 2nd packet ALWAYS conveys 8640 M altitude and BL for the char 5,6 grid. Voltage and temperature seem to be responding normal though.

Based on TELEN, this seems to only happen first in the morning, or, immediately after a afternoon reboot. (why its rebooting in the afternoon is another story). so based on this criteria perhaps this is something I can replicate on the ground (by forcing frequent reboots).

knormoyle, 2024-10-28:
my two cents:
I think having a test loop for gps cold reset would be useful

There are two aspects to reboot: rp2040 startup, and gps is starting with empty almanac

The ATGM336H-5N31 https://www.icofchina.com/d/file/xiazai/2016-12-05/b5c57074f4b1fcc62ba8c7868548d18a.pdf
only preserves the almanac if VBAT is between 1.5v and 3.6v

if you're expecting a gps cold reset because you disconnected power, well you usually have to wait 10-15 secs for caps to drain, otherwise VBAT may stay high enough so almanac is preserved.
it's easier to do repeated gps cold reset, because that flushes the almanac and all other gps data.

I'm assuming you don't force a gps cold reset on powerup, so the almanac would be preserved if VBAT is good

but wakeup in the morning on a balloon, and the almanac is definitely not there
Takes 2-3 minutes to load the almanac

U4B will send callsign and telemetry even if gps isn't valid

You should act like traquito and not send telemetry if the gps isn't valid. Don't use "the last one" in the morning (u4b does that..it can send a gps location from the prior day
 
but I really think you should have some kind of gps cold reset testing in the config (a loop)
lock time (which will include almanac time) can be grepped

If you have a noisy system, it might not be able to get the almanac ..the almanac has to come from a single satellite.
it's the hardest thing the gps system has to grab, in terms of data
these chips don't need to download a full almanac. they can get a fix with a partial almanac

knormoyle, 2024-10-28:
Oh: I was testing with a low voltage source. 
some questions:
Does your voltage supervisor on the jlcpcb  require input voltage >3v to boot? 
I'm not sure you've documented that?
what voltage will it cause reset at? (thinking of your afternnoon reboot. Sun is hot then, lower power/voltage from cells sometimes, because cells are hot

knormoyle, 2024-10-28:
I left a jlcpcb board running in my window last night
I tend to have a gps-challenging environment (u4b has a hard time with hardware test)..although traquito does fine with consistent 60 sec gps fixes (from cold reset)...with consistent altitude

I notice here the altitude changes from a correct 2358 meters to other values. Sat count varies, including going down to 0. highest is 15.  Note the series of wrong altitudes, including down to 1426 meters

it's cloudy, windy this morning
Also: curious why the Volts report says 3.1v ..is that the voltage the pico is seeing? (this is running off usb power)

00d08:26:37.048789 [14416] Temp: 84.2  Volts: 3.1  Altitude: 2358  Satellite count: 15
00d08:26:42.055336 [14417] Temp: 83.4  Volts: 3.1  Altitude: 2358  Satellite count: 15
00d08:26:47.061925 [14418] Temp: 83.4  Volts: 3.1  Altitude: 2358  Satellite count: 15
00d08:26:52.068457 [14419] Temp: 83.4  Volts: 3.1  Altitude: 2358  Satellite count: 15
00d08:26:57.075044 [14420] Temp: 81.7  Volts: 3.1  Altitude: 2358  Satellite count: 15
00d08:27:02.081584 [14421] Temp: 82.5  Volts: 3.1  Altitude: 2358  Satellite count: 15
00d08:27:07.088171 [14422] Temp: 84.2  Volts: 3.1  Altitude: 2359  Satellite count: 15
00d08:27:12.094713 [14423] Temp: 83.4  Volts: 3.1  Altitude: 2359  Satellite count: 15
00d08:27:17.101298 [14424] Temp: 83.4  Volts: 3.1  Altitude: 2359  Satellite count: 14
00d08:27:22.107845 [14425] Temp: 84.2  Volts: 3.1  Altitude: 2359  Satellite count: 14
00d08:27:27.114406 [14426] Temp: 82.5  Volts: 3.1  Altitude: 2360  Satellite count: 13
00d08:27:32.120966 [14427] Temp: 82.5  Volts: 3.1  Altitude: 2360  Satellite count: 14
00d08:27:37.127546 [14428] Temp: 83.4  Volts: 3.1  Altitude: 2360  Satellite count: 14
00d08:27:42.134090 [14429] Temp: 83.4  Volts: 3.1  Altitude: 2360  Satellite count: 14
00d08:27:47.140667 [14430] Temp: 82.5  Volts: 3.1  Altitude: 2360  Satellite count: 14
00d08:27:52.147203 [14431] Temp: 83.4  Volts: 3.1  Altitude: 2361  Satellite count: 14
00d08:27:57.153765 [14432] Temp: 83.4  Volts: 3.1  Altitude: 2361  Satellite count: 13
00d08:28:02.160312 [14433] Temp: 84.2  Volts: 3.1  Altitude: 2361  Satellite count: 13
00d08:28:07.166896 [14434] Temp: 82.5  Volts: 3.1  Altitude: 2361  Satellite count: 14
00d08:28:12.173441 [14435] Temp: 82.5  Volts: 3.1  Altitude: 2361  Satellite count: 14
00d08:28:17.180006 [14436] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:22.186555 [14437] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:27.193132 [14438] Temp: 82.5  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:32.199688 [14439] Temp: 82.5  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:37.206252 [14440] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:42.212771 [14441] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:47.219346 [14442] Temp: 82.5  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:52.225874 [14443] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:28:57.232467 [14444] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:29:02.239003 [14445] Temp: 83.4  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:29:07.245555 [14446] Temp: 80.0  Volts: 3.1  Altitude: 2362  Satellite count: 14
00d08:29:12.252079 [14447] Temp: 81.7  Volts: 3.1  Altitude: 2363  Satellite count: 14
00d08:29:17.258645 [14448] Temp: 82.5  Volts: 3.1  Altitude: 2363  Satellite count: 14
00d08:29:22.315805 [14449] Temp: 82.5  Volts: 3.2  Altitude: 2363  Satellite count: 14
00d08:29:27.322408 [14450] Temp: 85.9  Volts: 3.1  Altitude: 2361  Satellite count: 9
00d08:29:32.329001 [14451] Temp: 86.7  Volts: 3.1  Altitude: 2368  Satellite count: 7
00d08:29:37.335591 [14452] Temp: 77.5  Volts: 3.1  Altitude: 2372  Satellite count: 8
00d08:29:42.342216 [14453] Temp: 85.9  Volts: 3.1  Altitude: 2374  Satellite count: 7
00d08:29:47.348816 [14454] Temp: 82.5  Volts: 3.1  Altitude: 2375  Satellite count: 8
00d08:29:52.355389 [14455] Temp: 84.2  Volts: 3.1  Altitude: 2374  Satellite count: 8
00d08:29:57.361957 [14456] Temp: 86.7  Volts: 3.1  Altitude: 2365  Satellite count: 6
00d08:30:02.368517 [14457] Temp: 80.8  Volts: 3.1  Altitude: 2353  Satellite count: 7
00d08:30:07.375106 [14458] Temp: 83.4  Volts: 3.1  Altitude: 2341  Satellite count: 8
00d08:30:12.381696 [14459] Temp: 83.4  Volts: 3.2  Altitude: 2330  Satellite count: 8
00d08:30:17.388284 [14460] Temp: 80.8  Volts: 3.1  Altitude: 2318  Satellite count: 8
00d08:30:22.394879 [14461] Temp: 84.2  Volts: 3.1  Altitude: 2304  Satellite count: 8
00d08:30:27.401513 [14462] Temp: 84.2  Volts: 3.1  Altitude: 2292  Satellite count: 7
00d08:30:32.408132 [14463] Temp: 79.2  Volts: 3.1  Altitude: 2285  Satellite count: 8
00d08:30:37.414761 [14464] Temp: 86.7  Volts: 3.2  Altitude: 2286  Satellite count: 9
00d08:30:42.421355 [14465] Temp: 83.4  Volts: 3.1  Altitude: 2283  Satellite count: 8
00d08:30:47.427975 [14466] Temp: 84.2  Volts: 3.1  Altitude: 2368  Satellite count: 7
00d08:30:52.434579 [14467] Temp: 80.0  Volts: 3.1  Altitude: 2344  Satellite count: 8
00d08:30:57.441233 [14468] Temp: 86.7  Volts: 3.1  Altitude: 2343  Satellite count: 9
00d08:31:02.447865 [14469] Temp: 84.2  Volts: 3.1  Altitude: 2344  Satellite count: 6
00d08:31:07.454491 [14470] Temp: 88.4  Volts: 3.2  Altitude: 2342  Satellite count: 7
00d08:31:12.461123 [14471] Temp: 83.4  Volts: 3.2  Altitude: 2340  Satellite count: 8
[... cut ...]
00d08:41:13.361799 [14646] Temp: 84.2  Volts: 3.1  Altitude: 1693  Satellite count: 6
00d08:41:18.368367 [14647] Temp: 91.0  Volts: 3.1  Altitude: 1593  Satellite count: 6
00d08:41:23.425634 [14648] Temp: 83.4  Volts: 3.1  Altitude: 1523  Satellite count: 6
00d08:41:28.432252 [14649] Temp: 90.1  Volts: 3.1  Altitude: 1522  Satellite count: 6
00d08:41:33.438845 [14650] Temp: 86.7  Volts: 3.1  Altitude: 1487  Satellite count: 6
00d08:41:38.445427 [14651] Temp: 85.1  Volts: 3.1  Altitude: 1426  Satellite count: 6

knormoyle, 2024-10-28:
here's a sequence where altitude went bad but eventually recovered to good (it would be nice if lat/lon was printed in this messages also, to see if the horizontal fix went bad/good. Also: what is the first column? is utc timestamp on this somehow?

00d08:42:03.478330 [14656] Temp: 88.4  Volts: 3.1  Altitude: 1326  Satellite count: 5
00d08:42:08.484913 [14657] Temp: 85.9  Volts: 3.1  Altitude: 1262  Satellite count: 6
00d08:42:13.491481 [14658] Temp: 85.9  Volts: 3.1  Altitude: 1207  Satellite count: 6
00d08:42:18.498051 [14659] Temp: 86.7  Volts: 3.1  Altitude: 1157  Satellite count: 6
00d08:42:23.504646 [14660] Temp: 86.7  Volts: 3.1  Altitude: 1112  Satellite count: 5
00d08:42:28.511266 [14661] Temp: 84.2  Volts: 3.1  Altitude: 1072  Satellite count: 5
00d08:42:33.517876 [14662] Temp: 87.6  Volts: 3.1  Altitude: 1035  Satellite count: 5
00d08:42:38.524507 [14663] Temp: 86.7  Volts: 3.1  Altitude: 1002  Satellite count: 5
00d08:42:43.531129 [14664] Temp: 87.6  Volts: 3.1  Altitude: 971  Satellite count: 5
00d08:42:48.537780 [14665] Temp: 86.7  Volts: 3.1  Altitude: 948  Satellite count: 5
00d08:42:53.544395 [14666] Temp: 81.7  Volts: 3.1  Altitude: 948  Satellite count: 5
00d08:42:58.550991 [14667] Temp: 83.4  Volts: 3.1  Altitude: 949  Satellite count: 5
00d08:43:03.557614 [14668] Temp: 86.7  Volts: 3.2  Altitude: 951  Satellite count: 5
00d08:43:08.564244 [14669] Temp: 86.7  Volts: 3.2  Altitude: 954  Satellite count: 5
00d08:43:13.570865 [14670] Temp: 81.7  Volts: 3.2  Altitude: 934  Satellite count: 4
00d08:43:18.577450 [14671] Temp: 85.9  Volts: 3.1  Altitude: 921  Satellite count: 5
00d08:43:23.584023 [14672] Temp: 85.9  Volts: 3.1  Altitude: 898  Satellite count: 5
00d08:43:28.590612 [14673] Temp: 83.4  Volts: 3.1  Altitude: 898  Satellite count: 11
00d08:43:33.597170 [14674] Temp: 83.4  Volts: 3.1  Altitude: 2372  Satellite count: 13
00d08:43:38.603730 [14675] Temp: 85.9  Volts: 3.1  Altitude: 2370  Satellite count: 13
00d08:43:43.610295 [14676] Temp: 83.4  Volts: 3.1  Altitude: 2370  Satellite count: 12
00d08:43:48.616874 [14677] Temp: 83.4  Volts: 3.1  Altitude: 2370  Satellite count: 12
00d08:43:53.623418 [14678] Temp: 85.1  Volts: 3.1  Altitude: 2370  Satellite count: 13
00d08:43:58.630011 [14679] Temp: 84.2  Volts: 3.1  Altitude: 2371  Satellite count: 13
00d08:44:03.636573 [14680] Temp: 82.5  Volts: 3.1  Altitude: 2372  Satellite count: 14
00d08:44:08.643148 [14681] Temp: 83.4  Volts: 3.1  Altitude: 2372  Satellite count: 14
00d08:44:13.649706 [14682] Temp: 84.2  Volts: 3.1  Altitude: 2372  Satellite count: 14

knormoyle, 2024-10-28:
Don't know if this is expected or maybe related. Note the loss of 3d fix while sitting there. Why would that happen if the tracker isn't moving?

00d08:33:07.671481 [14536] Temp: 86.7  Volts: 3.1  Altitude: 2396  Satellite count: 6
00d08:33:12.678101 [14537] Temp: 81.7  Volts: 3.1  Altitude: 2396  Satellite count: 7
00d08:33:13.178993 [14538] Gps was available, but no valid 3d Fix. ledmode 3 XMIT status 1
00d08:33:13.679837 [14539] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:14.180660 [14540] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:14.681474 [14541] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:15.182283 [14542] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:15.683089 [14543] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:16.183908 [14544] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:16.684723 [14545] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:17.185534 [14546] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:17.686345 [14547] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:17.686494 [14548] Temp: 85.9  Volts: 3.1  Altitude: 2396  Satellite count: 0
00d08:33:18.187424 [14549] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:18.688236 [14550] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:19.189039 [14551] Gps was available, but no valid 3d Fix. ledmode 4 XMIT status 1
00d08:33:22.693712 [14552] Temp: 84.2  Volts: 3.1  Altitude: 2388  Satellite count: 7
00d08:33:27.700303 [14553] Temp: 84.2  Volts: 3.1  Altitude: 2378  Satellite count: 12
00d08:33:32.706860 [14554] Temp: 82.5  Volts: 3.1  Altitude: 2371  Satellite count: 12
00d08:33:37.713446 [14555] Temp: 85.9  Volts: 3.1  Altitude: 2367  Satellite count: 12
00d08:33:42.719979 [14556] Temp: 85.9  Volts: 3.1  Altitude: 2367  Satellite count: 12

EngineerGuy314, 2024-10-29:
>>Why would that happen if the tracker isn't moving?
it was only receiving 6 or 7 satellites before that.  The tracker wasn't moving, but the sats were :-)   but in general, i think only seeing 6 or 7 sats means you are very close to the edge of not enough satellite signal. indoors, and also changing cloud coverage can really attentuate the sat signals.

>>Does your voltage supervisor on the jlcpcb require input voltage >3v to boot?

i believe the device i used has a threshold of 3.3v, and no i dont think i specifically documented that. That is kindof an arbitrary level, and further experimentation could be done to possibly use a lower value.

Also, you said it was reporting 3.1 volts when plugged into USB. that should not be, it should have reported close to 5. Unless you had a defective usb cable or port, that should be a new issue.

EngineerGuy314, 2024-10-29:
kevin, based on our email exchange i am pretty sure this particular issue is caused by incorrectly parsing GPS messages that dont have all the expected fields, like you mentioned. I will fix shortly...

knormoyle, 2024-11-01:
> Also, you said it was reporting 3.1 volts when plugged into USB. that should not be, it should have reported close to 5. Unless you had a defective usb cable or port, that should be a new issue.

the code was doing wrapping in main.c before that print
USB voltage of 5.0v is not in the 3-4.95v range that u4b supports without wrapped
So any high USB voltage could cause a wrap (in telemetry)
I changed that in my fork so that the modulo stuff for temp and voltage doesn't happen in main where it prints to stdout

knormoyle, 2024-11-01:
> kevin, based on our email exchange i am pretty sure this particular issue is caused by incorrectly parsing GPS messages that dont have all the expected fields, like you mentioned. I will fix shortly...

I added some enhancements to my fork in gpstime.c to have the parsing be more obviously robust in the present of empty fields etc. (the variable length char strings and the way commas were turned into null terminators for char extraction is a little nonintuitive.
you can look at what I did

but I compared GNGGA sentences and altitude extractions, and I don't believe there was a problem there.
the code I changed just is a little more understandable that it covers the empty cases, I think (or not enough commas, or some other error cases I think)

knormoyle, 2024-11-01:
> i believe the device i used has a threshold of 3.3v, and no i dont think i specifically documented that. That is kindof an arbitrary level, and further experimentation could be done to possibly use a lower value.

in general, for LDO's the input voltage has to be equal to the target regulated voltage + the drop voltages.
So I think Vin needs to be around 3.55v

3.3v will pass thru, but it won't be regulated as well, if I remember correctly.

EngineerGuy314, 2024-11-05:
a while back i started to "snapshot" position and altitude when the first packet is created, and use those saved values for the 2nd telemetry packet (otherwise strange things happened if major grid changed during 1st/2nd packet). In the event that the pico boots up too late to create the 1st packet, but in time to create the 2nd, the code generates the 2nd packet with data from the "snapshot" variables, which contained bogus data because the first packet code never ran and never wrote data into them. now fixed.

closed 2024-11-05 by EngineerGuy314