EngineerGuy314/pico-WSPRer #49: is there no way to recover from 'too high" Klock clock setting

knormoyle, 2024-10-31:
I accidently set K to 260 in the config menu
when I started putty again, it had trying setting the clock to 260 but then hung

it seems like since the 260 is in the nvram, it will always try to set that frequency when I reboot

reloading the uf2 doesn't clear the clock freq to lower default.

hmm. I think I can load a nother uf2 that will wipe the nvram
will that work?
https://github.com/dwelch67/raspberrypi-pico/blob/main/flash_nuke.uf2

installing that fixed things. It wiped the nvram and while I couldn't putty, it put me back into folder to load a new copy of the pico-WSPRer.uf2..so I no longer had the erroneous high frequency

knormoyle, 2024-10-31:
then when I started it noted the bad values in nvram now

***********
NVRAM dump:
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

BAD values in NVRAM detected! will reboot in 10 seconds... press any key to enter user-setup menu..
all the values were bad, so I had to manually set everything.

I suppose it might be nice if there was a single command to everything to default values, to recover from this situation
maybe not important.

maybe not important..just need to document what to do in this case (bad Klock)
~

knormoyle, 2024-10-31:
here's a more official flash nuke. I've not tried this one, so don't know if it loads with zeroes

Resetting Flash memory
Pico’s BOOTSEL mode lives in read-only memory inside the RP2040 chip, and can’t be overwritten accidentally. No matter what, if you hold down the BOOTSEL button when you plug in your Pico, it will appear as a drive onto which you can drag a new UF2 file. There is no way to brick the board through software. However, there are some circumstances where you might want to make sure your Flash memory is empty. You can do this by dragging and dropping a special UF2 binary onto your Pico when it is in mass storage mode.

Download the [UF2 file] 
https://datasheets.raspberrypi.com/soft/flash_nuke.uf2

See the [code on Github]
https://github.com/raspberrypi/pico-examples/blob/master/flash/nuke/nuke.c

knormoyle, 2024-11-01:
I did a bunch of things in using the pico sdk for checking whether it's going to be successful with a particular voltage before taking it, and fixed up some things for recovering when the nvram has all bad values (like all 1's) from a flash nuke to recover from a bad (too high ) Klock speed setting

I think everything is recoverable now, or shouldn't happen normally. If you set it to something like 400 and it wedges, you'll have to nuke the nvram with another uf2 (available online) but then even if the nvram is all 1's, it self recovers to a default clock (115) but you'll have to manually set some other things
mainly callsign, channel and importantly the custom pcb =1 (or it will look like you can't talk to gps

EngineerGuy314, 2024-11-01:
for the first several seconds the clock is always supposed to run at default speed.
the way i intended it to work, and the way it worked for me when i also bricked a pico with too high a clock frequency, is that you have a few seconds after power-up to enter the user-interface menu. clock speed is supposed to remain default until after the user-menu is closed. 

but i may have seen this same thing with a v2 pcb, so i will leave this open and have to look into it. (i also used the nuke file, among many other things i tried. did eventually recover).

after running the nuke, nvram is obvioulsy wiped and so its expected to give you that "BAD values in NVRAM detected! will reboot in 10 seconds... press any key to enter user-setup menu.."

and the reason i force a reboot after 10 seconds is in case there is a mis-read nvram during a flight. it will reboot and try reading nvram again until the data looks correct

EngineerGuy314, 2024-11-01:
so i just tried this with one of my v2 PCBs. i set Klock to 270Mhz (which i knew would lock up this batch) and it did lock. after power cycling it would blink for a few seconds and then lock up again.

I unpluged and re-plugged to the PC, and this time I connected Putty and hit a key ASAP and it opened the setup menu and let me change the Klock back down. 

I will close this issue for now until it can be replicated. you just need to be quick when trying to open the setup menu.

closed 2024-11-01 by EngineerGuy314