EngineerGuy314/pico-WSPRer #55: Wrong frequencies

IU5HKU, 2024-11-24:
Hi, i'm testing the multiband code, but unluckly for me, it doesn't work as expected. The only frequency who lies in the small wspr tx window is the 20m one, all the others are out of frequency of an amount who depends on the choosen frequency. 
I'm using a RP pico one, with 2040  mcu. I've sets 'X' as suffix and, '--' as Ub4 all the remaining settings are left as default. What i'm doing wrong? Ah, when the tx starts, it's not wspr-modulated, in the waterfall it looks like the 'test frequency', and, as said before, out of the wspr window by some kzh...
Thank you for your time
 MC

EngineerGuy314, 2024-11-25:
You can't run bands higher than 20M without overclocking.  For instance for 10M you would need to set the Klock param to 230Mhz at least. You could probably do 17M at 150Mhz, but I haven't tried. Unfortunately not every Pico likes to be overclocked, so you need to try it and find out.

Also, If you did not wire up the PPS signal as shown (or if the GPS does not have a position fix) the output frequency will be uncalibrated and likely very wrong.

If still no luck, try the non multiband version and see if that behaves the same. Let me know, thanks. And please try to include a screenshot of the user-config screen.

Using the semi-secret standalone beacon mode ('X' as suffix and, '--' as Ub4) has not been tested on multiband. If it works on the regular version but not multiband I need to dig into that a bit...

IU5HKU, 2024-11-25:
Yes, now everything works as expected, my fault was to not overclock the mcu. Values you told me are correct, 150Mhz at least for 17m, not less than 230Mhz for 10m, my pico board (the original one) support without effort an overclock till 250Mhz, i'll try if it still works with a cheap chinese clone asap. I'm using the pps, yes, i've followed the circuit you show in the wiki page. I confirm that 'X' and Ub4 '--' works without problems in the multiband version, well done.
Please, add this useful hints about frequencies and overclocking in the wiki page, so other users can benefit about that. 
I'm also using an 1:9 balun, with rf output pins practically in short circuit, without overheating problems, as you told me in the past.
My compliments for this project, clean code and works like a charm.

closed 2024-11-25 by EngineerGuy314

EngineerGuy314, 2024-11-25:
This got me thinking, so I put a scope on the 3v3 and looked at behavior during startup with different Klock speeds. Previously, I was setting the clock speed, then enabling the GPS unit. When the GPS unit is first enabled it makes the supply sag to about 2.9v for 0.1mS. It turns out the rp2040 can tolerate that sag easily normally, but not when overclocked. 

So I just changed it to power up the gps first, wait a couple milliseconds, and only then set the clock speed higher. Now I can run a full 270Mhz on the same boards that wouldn't work before! Change has been pushed to repo.