EngineerGuy314/pico-WSPRer #60: pre compiled "normal" Version end in Boot Loop

dl1esk, 2025-04-09:
hi there,

today i tried to flash the pre compiled version on my pico and it ends in a boot loop...
only the pre compiled multiband version works..
i also compiled it by myself and that version also doesnt work...

superkuh, 2025-04-10:
By boot loop do you mean it never presents the serial interface or anything and just constantly boots? Which file specifically? https://github.com/EngineerGuy314/pico-WSPRer/raw/main/build/pico-WSPRer.uf2 ? 

I flashed the above linked pre-compiled version yesterday and it boots and stays running long enough for me to use the serial port interface to configure it. But it did reboot randomly a few times during the process.

dl1esk, 2025-04-11:
Yes i mean this Version.... i tried it on 2 Raspi Pico`s ordered from Reichelt...
i couldn`t connect to the Serial Interface because it reboots always aprox every 20 Seconds... The Com Port was shown in Windows..
the multiband version works fine...

EngineerGuy314, 2025-04-11:
What is the behavior of the LED during the bootloop?

If the program detects corrupted data in the NVRAM (which it certainly will the first time you use a new pico) it reboots after a few seconds, but you should have just enough time to connect to the serial port and hit a key to initiate the user-config menu.

If any of the values look corrupted (ie weird characters in the DEXT config) then re-configure the offending values.  I think when i changed the extended telemetrt config from 4 characters to 3 it reboots unless you explicitly put some value into the DEXT config (three dashed --- if you want DEXT disabled)

dl1esk, 2025-04-11:
it`s a RP2040 based Pico... 
When it starts the LED is blinking quick for about 5 Secs after that the LED is pulsing bright to dark for about 10 Seconds then you hear the reconnect Tone on Windows and it is restarting...

dl1esk, 2025-04-11:
i could connect !!
BAD values in NVRAM detected! will reboot in 10 seconds... press any key to enter user-setup menu..
but why it doesnt do that on the multi band version ?!

EngineerGuy314, 2025-04-12:
>>but why it doesnt do that on the multi band version ?!
the multiband version (which was created by someone else) hasn't been maintained and is based on a very old version which probably doesn't include the nvram corruption testing.

>>i could connect !!
If you could please confirm so that I can close this issue: after you connect, if you go through each and every setting in the menu and enter valid data, is it then able to boot normally?

(if not, please post screenshot of what you see in the user menu).

Thanks

superkuh, 2025-04-13:
It is not 20s for me. It reboots itself exactly every 70 seconds (1 min 10s). It is very reliable. And each time the TTY name changes so I have to go back and type again in my serial terminal program.

Will it be rebooting itself every 70s when it is on battery an in operation too? Or is this just a "when connected to usb->serial" sort of thing? Or is it unintentional and a crash like this issue suggested?

dl1esk, 2025-04-13:
yes when i change the data then it boots normally...
why does the NVRAM Check not just Stop with a notice to change to valid data so that it doesn`t reboot everytime

whallmann, 2025-04-13:
It does wait long enough to hit any key. I just programmed the non
multiband version, and it shows the same known behaivor as anny other days
befor
73 Wolfgang DF7PN

EngineerGuy314, 2025-04-13:
>>why does the NVRAM Check not just Stop with a notice to change to valid data so that it doesn`t reboot everytime

Excellent question, and originally that's how it worked. But, some Raspberry Picos have been known to intermittently read some bad data on startup, especially with temperature and voltage fluctuates wildly.

If a balloon wakes up at altitude when the sun rises and it reads bad NVRAM data, then you will lose an entire day of transmission.  But with current behavior it will reboot itself and try to read the NVRAM again. It repeats that forever until hopefully it reads good data, and then begins transmitting.

EngineerGuy314, 2025-04-14:
@dl1esk  i am assuming this is resolved. If not please open a new issue

closed 2025-04-14 by EngineerGuy314