EngineerGuy314/pico-WSPRer #61: TXing supresses gps reception?

whallmann, 2025-04-13:
Hi friends

today I used the last provided uf2 file for the raspi pico.
Thats the one who uses the antenna port near the resistor network for Startup.  
I wounder why the traquito status page for my call df7pn on 493 channel shows null satellite  on KNOTS column. Other dsta like qra locator is available.

A look on the serial output tells me, that there is an satcounter more than 10.  Just in the moment the TX goes around the fix gets lost and the counter jumps to zero. 

Is this a new feature to save gps power or is the HF power blocking the gps receiver now?

cu Wolf df7pn

EngineerGuy314, 2025-04-13:
>>Is this a new feature to save gps power or is the HF power blocking the gps receiver now?

not a new feature, the HF power is overloading the GPS input, which is a common and known problem. I added an entire section to the wiki on how to mitigate this problem: 

https://github.com/EngineerGuy314/pico-WSPRer/wiki/pico%E2%80%90WSPRer-(aka-Cheapest-Tracker-in-the-World%E2%84%A2)#GPS-loss-mitigation

I have a flight up now where the only mitigation I used was a 94mm dipole for gps input (instead of a monopole).  Extended telemetry shows that it only momentarily loses GPS lock maybe once every 7 days. Reports on average about 9 or 10 satellites used for fix, occasionally down to 5, sometimes 14.

Testing on the ground will always show much much worse performance because of atmosphere and obstructions. Not a concern for flight.

Now, one thing that I have never tested (and this is partly why I am not interested in creating a multiband version) is if transmitting on bands other 20M create worse GPS interference.  Maybe, maybe not, but I can't vouch for it. 

Read the section on GPS loss mitigation, and I very very strongly recommond:` "DEXT telemetry configuration of "0-0" is very effective"`

This way there is no transmission during the 4th slot, which will always allow GPS to re-acquire lock and calculate frequency correction. The fifth will be "type 0 DEXT", and worst case if you always lose GPS lock during TX the frequency will drift and you won't get a read on slot 5. BUT, by the time it then starts regular WSPR on slot 0 the chip temperature and frequency drift has stabilized and you have a very good chance of reception of WSPR, standard telemetry, and extended telemetry in slot 3. Basically, this makes sure to pre-warm the chip before the important slots.

whallmann, 2025-04-14:
Oh yeah, Thanks Rob for the additional informations.
I worked on it and it works now with the better gps dipole.  The reduction
of the sat count during transmit is still visible but it hurts not so much
as before.  Also the change of DEXT to recommended 0-0 is a good hint.
So thanks a lot

closed 2025-04-14 by EngineerGuy314