EngineerGuy314/pico-WSPRer #63: Question's

edegraaff, 2025-05-21:
Hi, i am currently using your project as a wspr transmitter for some testing purposes like antennas but. I am also interested in HAB's. I would build your project with solar panels and launch a floater in the future. I was wondering in the documentation is mentioned the option to transmit on 10 , 20 and 40 m something i think is nice to have. I wanted to activate this option but i have no clue if it is in the current build or not. 
"secret band Hopping mode: not fully tested yet, and has many limitations. It transmits U4B packets on both 20M and 10M"
Since it will be the first flight with a pi pico i want to build it is basic as possible. U have done some flights with RS41's.
For the floater where do you get your balloons? 
Do you have some practical tips? 
Regards
Eelco de PD1EG

EngineerGuy314, 2025-05-22:
I have had the best luck with "Orbs 3" Clear Pico Balloon" from balloons.online. A lot of others have had success with PartyWoo balloons from Amazon. I strongly suggest you join the group at https://groups.io/g/picoballoon there is a wealth of practical info there. Nearly everyone is very helpful and friendly with newcomers. There is a lot of information to digest and a lot of new equipment and skills that might be needed for your first pico launch, but thats what makes it so satisfying to pull one off. 

The "official" firmware does have a band-hopping mode to alternate between 10 and 20M. I never tested it in the air  because I realized its not easy making an antenna that works good on both those bands without traps, etc. There is also the "multi-band" version, but that is not actively supported and uses a cruder method for RF generation.

You can change the frequencies used on lines 127 to 130 of Main.c to whatever band you want. I included thorough instructions on the wiki to easily recompile the project yourself, all you need is a Raspberry Pi or any other linux system. So please feel free to experiment.

I can confidently say that 20 Meters on this system works great, and i have numerous flights and circumnavigations to confirm that. But I cannot say the same about other bands, so that is why I am hesitant to release an all-band version.  Because of the novel way that the code forces the chip's PLL to generate RF, there are sometimes scenarios that cause problems.

edegraaff, 2025-05-22:
Hi Thanks for your reply, based on that i will use the official one band version and tune the 20 meter band with the right lenght of copperwire for 20 meter. If i have the device ready i could test the bandhopping and see if it will add some value. I also need to find the right balloon here in the Netherlands i will post a question in groups.io regarding this.

EngineerGuy314, 2025-05-22:
Hi Eelco,

Hope I catch you i time, but I just pushed a change that theoretically adds band selection to the main program. It is completely untested, but "should" work. Please let me know how it is.

EngineerGuy314, 2025-05-22:
Please create a new issue if there are any problems with it, thanks.

closed 2025-05-22 by EngineerGuy314