EngineerGuy314/pico-WSPRer #66: is your software compatible to traquito tracker Hardware?

whallmann, 2025-08-01:
I would prefere to use your well known Software on the traquito trackers (hat for pico). I think Doug is doing the same as yours pico source. Did you already checked this Rob? Until now the boards are not printed yet 😉 if i need to get it run on this hardware some changes are possible.

cu Wolf DF7PN

EngineerGuy314, 2025-08-01:
Hello, no its not compatible. The traquito uses a Si5351 for the RF gen. 

Kevin AD6Z has a similar tracker as mine, with similar open source C code, but his does use the Si5351.  I suggested to him that with minor modifications he can make his work with Traquito hardware.

EngineerGuy314, 2025-08-03:
you can also check out: github.com/IveyWorks/OpenTJP

its a completely open source program compatible with Traquito hardware